scsd-cdh/OBC #38: CDH Board

silvehh, 2025-02-23:
also CDH Board

m0Zahed, 2025-03-02:
[image]

Below is a **step-by-step** demonstration of how you might organize tasks in FreeRTOS for each subsystem shown in your diagram (AstroYeast, Transceiver, Backplane, PDS, BMS, ADCS), with the **CDH SAMV71** acting as the central coordinator. This includes:

1. **Sample Global State Variables** (wrapped in a struct).  
2. **Synchronization** using a **mutex** (for safe access to global state) and optionally **queues** (for message passing).  
3. **Function Declarations** of tasks (not full implementations) to show how you’d structure your code.

We’ll finish with a **detailed tutorial** on setting this up in a FreeRTOS project, plus discussion on whether to use **message queues** or **global state variables** for inter-task communication.

---

## 1. High-Level Architecture

In your diagram, the **CDH SAMV71** updates or coordinates “global state variables” for each subsystem:
- **AstroYeast** (`AY_SM`)
- **Transceiver** (`TCV_SM`)
- **Backplane** (`BP_SM`)
- **PDS** (`PDS_SM`)
- **BMS** (`BMS_SM`)
- **ADCS** (`ADCS_SM`)

Each subsystem can be represented as a **FreeRTOS task**. We can store shared status information in a global data structure (protected by a **mutex**). Alternatively, tasks can communicate via **queues** or **direct-to-task notifications**.

---

## 2. Sample Global State Variables

Create a single global structure that holds relevant states for each subsystem. For instance:

```c
/* GlobalState.h */

#ifndef GLOBAL_STATE_H
#define GLOBAL_STATE_H

#include <stdint.h>
#include <stdbool.h>

/* Example states for each subsystem. Adjust fields as needed. */
typedef struct
{
    /* AstroYeast subsystem state */
    bool  ay_active;
    uint8_t ay_experimentStage;

    /* Transceiver subsystem state */
    bool  tcv_linkActive;
    uint32_t tcv_bytesSent;
    uint32_t tcv_bytesReceived;

    /* Backplane subsystem state */
    bool  bp_isHealthy;
    uint16_t bp_errorCode;

    /* PDS subsystem state */
    float  pds_batteryVoltage;
    float  pds_batteryCurrent;
    bool   pds_powerGood;

    /* BMS subsystem state */
    bool   bms_charging;
    bool   bms_fault;
    uint8_t bms_tempSensorCount;

    /* ADCS subsystem state */
    bool   adcs_enabled;
    float  adcs_pitch;
    float  adcs_yaw;
    float  adcs_roll;
[... 233 lines cut ...]

Often, **hybrid** solutions are used:
- A **global state** for overall status (like “BMS fault = true/false”).  
- **Message queues** for event-driven data (like “new sensor reading,” “command from ground,” etc.).

---

## 8. Making It “Super Nice”

To keep your code well-organized and robust:

1. **Modularize**  
   - Put each subsystem’s task in its own source file (e.g., `AstroYeast.c`, `Transceiver.c`), with a matching header for function declarations.  
   - Have a `GlobalState.c/h` or a “SystemManager.c/h” that coordinates the main data structure.

2. **Document**  
   - Add Doxygen comments or other inline documentation describing each field in the global struct, each task’s purpose, and any concurrency rules.

3. **Use FreeRTOS Features**  
   - Consider **event groups** if you have multiple flags that tasks wait on.  
   - Use **direct-to-task notifications** if you want to avoid the overhead of queues for small events.

4. **Plan Priorities**  
   - Time-critical tasks (like real-time sensor reading) might have higher priority.  
   - Housekeeping tasks can have lower priority.

m0Zahed, 2025-03-02:
Each subtask will be triggered upon receiving an IRQ from that subsystem.

**IRQs are raised to request fetching updates from each board. Here is an interface diagram.**

<p align="center">
  [image]
</p>